mriscoc/Marlin #57: G35 command fails

AshenColors, 2021-07-03:
I'm currently using the latest beta (20210623) for BLTouch 4.2.2 on my Ender 3v2, which I've reset to default settings. When connected through Pronterface, the G35 Assisted Tramming command fails after the initial center z-probe from homing. I've verified that the x-y offset for the BLtouch is set properly, but that doesn't seem to help. All other leveling functionality appears to work as intended.

```>>> g35
SENDING:G35
G35 failed at point 0 (Front-Left) X29.00 Y29.00
G35 aborted.
ok P15 B3```

AshenColors, 2021-07-03:
Already figured it out — it's trying to apply a X-46 offset because of how the BLTouch is configured (M851
Probe Offset X-46.00 Y-8.00), fails because it can't go that far to get the probe in the correct position, and gives up. I don't *think* this is a firmware problem, or at the very least, it's also upstream's problem.

AshenColors, 2021-07-03:
The main Marlin repo has a sanity check for the tramming points being valid in respect to the probe-nozzle offset at compile time, but not at run time. I've verified the 'fix' on the main repo, but I am unable to successfully compile this fork to test the config since platformio complains about a "disk_io_driver.h" being missing, which is its own problem entirely, so I'm closing this out as "filed a little too early".

closed 2021-07-03 by AshenColors

mriscoc, 2021-07-06:
The G35 needs that the BLTouch can reach the points where the screws are physically positioned on the bed.