mriscoc/Marlin #60: This is based on GPL v3.0 code. You MUST disclose the source.

strayr, 2021-07-08:
### Description

No Source PROVIDED

### Steps to Reproduce

view repo

**Expected behavior:** 

Source included

**Actual behavior:** 

Source deliverately not included

#### Additional Information

From https://github.com/MarlinFirmware/Marlin/blob/2.0.x/LICENSE

                    GNU GENERAL PUBLIC LICENSE
                       Version 3, 29 June 2007

 Copyright (c) 2007 Free Software Foundation, Inc. <https://www.fsf.org/>
 Everyone is permitted to copy and distribute verbatim copies
 of this license document, but changing it is not allowed.

                            Preamble

  The GNU General Public License is a free, copyleft license for
software and other kinds of works.

  The licenses for most software and other practical works are designed
to take away your freedom to share and change the works.  By contrast,
the GNU General Public License is intended to guarantee your freedom to
share and change all versions of a program--to make sure it remains free
software for all its users.  We, the Free Software Foundation, use the
GNU General Public License for most of our software; it applies also to
any other work released this way by its authors.  You can apply it to
your programs, too.

  When we speak of free software, we are referring to freedom, not
price.  Our General Public Licenses are designed to make sure that you
have the freedom to distribute copies of free software (and charge for
them if you wish), that you receive source code or can get it if you
want it, that you can change the software or use pieces of it in new
free programs, and that you know you can do these things.

  To protect your rights, we need to prevent others from denying you
these rights or asking you to surrender the rights.  Therefore, you have
certain responsibilities if you distribute copies of the software, or if
you modify it: responsibilities to respect the freedom of others.

  For example, if you distribute copies of such a program, whether
gratis or for a fee, you must pass on to the recipients the same
freedoms that you received.  You must make sure that they, too, receive
or can get the source code.  And you must show them these terms so they
know their rights.

  Developers that use the GNU GPL protect your rights with two steps:
(1) assert copyright on the software, and (2) offer you this License
giving you legal permission to copy, distribute and/or modify it.

  For the developers' and authors' protection, the GPL clearly explains
that there is no warranty for this free software.  For both users' and
authors' sake, the GPL requires that modified versions be marked as
changed, so that their problems will not be attributed erroneously to
authors of previous versions.

  Some devices are designed to deny users access to install or run
[... 599 lines cut ...]
Also add information on how to contact you by electronic and paper mail.

  If the program does terminal interaction, make it output a short
notice like this when it starts in an interactive mode:

    {project}  Copyright (c) {year}  {fullname}
    This program comes with ABSOLUTELY NO WARRANTY; for details type `show w'.
    This is free software, and you are welcome to redistribute it
    under certain conditions; type `show c' for details.

The hypothetical commands `show w' and `show c' should show the appropriate
parts of the General Public License.  Of course, your program's commands
might be different; for a GUI interface, you would use an "about box".

  You should also get your employer (if you work as a programmer) or school,
if any, to sign a "copyright disclaimer" for the program, if necessary.
For more information on this, and how to apply and follow the GNU GPL, see
<https://www.gnu.org/licenses/>.

  The GNU General Public License does not permit incorporating your program
into proprietary programs.  If your program is a subroutine library, you
may consider it more useful to permit linking proprietary applications with
the library.  If this is what you want to do, use the GNU Lesser General
Public License instead of this License.  But first, please read
<https://www.gnu.org/licenses/why-not-lgpl.html>.

mriscoc, 2021-07-09:
This forks PR to the main Marlin repository, the up-to-date PR (https://github.com/MarlinFirmware/Marlin/pull/21942) can be found here: https://github.com/mriscoc/Marlin_Ender3v2/tree/bugfix-2.0.x-DWIN-upgrade

closed 2021-07-09 by mriscoc

mriscoc, 2021-07-13:
Be happy, and wait for the approbation of the pull requests in the official Marlin repository.
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

tititopher68-dev, 2021-07-13:
@tititopher68-dev 

Your responses and your practice of removing what bothers you, are ridiculous and frankly hurt the community. You pretend to be the God of programming, but you share nothing at all or very little ! I asked you a question in the Chat area and the only thing you did is delete that question!

Clearly answer this question :

You published your last Release # 45 with :
Filament run-out sensor with active state setup
Transparent icons for better color theme adaptability
Lock screen
Editable physical dimensions
etc ...
You wrote :
**When this version gets tested enough is renamed to BETA and is available to be tested by anyone, at this time, the previous BETA is released as Ender3v2. For compile yourself, grab the current source files then apply the correct configuration files from the configurations folder.**

You wrote the same things on Telegram !

BUT it's not true! Nothing in the sources version Ender3V2-Released or DWIN ....

A little more honesty in your announcements and in your responses. You drowned the fish even when you are officially asked to be clear in your sources ... (request that you close immediately!) ....

tititopher68-dev, 2021-07-13:
@strayr 
The only response from this "famous" developer Mrisco was to remove the disturbing questions about these sources !
It does not respect the GPL3.0 in any way and I have kept track of what I say.
He's hiding behind pseudo PR on Marlin's Github, but he's not. We are several to have searched but without any trace of what he says. On the other hand, to take up the work of some, it does it without worries and without citing the authors!

mriscoc, 2021-07-13:
The complete version: 
**Remarks**
Please remember that it is a preliminary and bugfix release be careful. If you are an advanced user, you can get the last experimental version at the Telegram firmware group: https://t.me/Ender3v2_Firmware where we test the firmware and has new features added. When this version gets tested enough is renamed to BETA and is available to be tested by anyone, at this time, the previous BETA is released as Ender3v2. For compile yourself, grab the current base source files then apply the correct configuration files from the configurations folder, **most of the more recent features or additions will be released through pull requests to the official Marlin repository**.