mriscoc/Marlin #63: Include Meatpack

thomasvt1, 2021-07-11:
Hi,

Thanks for this great firmware! Altough this firmware uses a higher baud rate to communicate with the printer. Wouldn't using Meatpack also double the baud rate since in most cases it compresses the dataq bij 50%?

https://github.com/scottmudge/OctoPrint-MeatPack/

Thanks for your time!

mriscoc, 2021-07-12:
Hi, more efficient is to double the baud rate, Meatpack can increase the error rate at that speed. Meatpack is not compatible with the binary file transfer protocol (BFT) that we use to update the firmware. Therefore, if we activate Meatpack, we must to decrease the communication speed and will not had the guarantee of get a 50% of compression, also we will lose the BFT compatibility.

closed 2021-07-12 by mriscoc

thomasvt1, 2021-07-12:
Makes sense! Thanks