mriscoc/Marlin #73: BLtouch Doesn't Level Correctly / Failed Probing

WheezerX, 2021-07-21:
### Description

BLTouch fails probing

### Steps to Reproduce

1. Installed Beta-422-BLTouch-20210714-143239.bin
2. Changed Phyiscal size to my printers correct dimensions
3. Attempted Leveling with BLtouch
4. BLtouch touches center fine, but when attempting the first point, it doesn't lower completely
5. Changed size back to stock
6. Attempted Leveling with BLtouch
7. Same as 4
8. Installed stock firmware
9. Attempted leveling with Bltouch
10. Successfully completed
11. Reinstalled Beta-422-BLTouch-20210714-143239.bin
12. Attempted leveling with BLTouch
13. Same as 4

**Expected behavior:** Expected leveling feature to... work.

**Actual behavior:** The BLTouch "needle" barely touches the first point on the bed, retracting maybe 1mm and pulls up completely and reports a failed probing after 3 attempts. Interestingly, it does successfully touch the center most point when homing.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
I used the precompiled Beta-422-BLTouch-20210714-143239.bin
* Provide pictures or links to videos that clearly demonstrate the issue.
This is a video I made showing the issue: https://youtu.be/Z9HqtCGPg98
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

madorax251, 2021-07-22:
I tried fw mriscoc and experienced the same thing. in the end I went back to using jyers fw 1.3.4 I suggest you use jyers firmware only because mriscoc only imitates fw jyers and only modifies it

WheezerX, 2021-07-28:
Issue resolved. Nylon Lock Nut removal from underside of bed resolved the issue completely. User 10 D 10 T error

closed 2021-07-28 by WheezerX