mriscoc/Marlin #76: Wrong temperatures when resuming paused print

AdamWP, 2021-07-24:
When pausing, purging, and resuming a print the nozzle temperature dropped to 185 and bed to 0 and did not come back up to the previous temperature.

njmV, 2021-07-29:
Same here. Temperature drops to the minimum extrude temperature of 185º and does not resume to the previous printing one. I always have to manual “fix it”.

mriscoc, 2021-08-01:
Thank you for your report, I will to investigate.

CyJaxX, 2021-08-05:
To reproduce this problem:
During printing, select pause and then confirm. Wait for printer to start cooling, then click to heat and resume printing. The nozzle temp will be set at 185 and the bed at 0.
The problem also occurs if the advanced pause is triggered by the filament runout sensor.

mriscoc, 2021-08-05:
> To reproduce this problem:
> During printing, select pause and then confirm. Wait for printer to start cooling, then click to heat and resume printing. The nozzle temp will be set at 185 and the bed at 0.
> The problem also occurs if the advanced pause is triggered by the filament runout sensor.

There is an Advanced pause (Change Filament) on Tune menu.

mriscoc, 2021-08-08:
> When pausing, purging, and resuming a print the nozzle temperature dropped to 185 and bed to 0 and did not come back up to the previous temperature.

The last experimental version is under test and fixed that issue.

closed 2021-08-12 by mriscoc