mriscoc/Marlin #97: Custom Giadej interface icons not appearing after flashing DWIN_SET to LCD

josurr, 2021-09-06:
### Description

The interface icons (and also parts of the menu)

### Steps to Reproduce

1. Download .zip of the main mriscoc/Marlin_Ender3v2 codebase
2. Extract .zip
3. Navigate to "Display assets" folder and open it
4. Open the "Giadej compilation" folder
5. Copy the DWIN_SET folder to a microSD card (formatted FAT32, 4096 byte allocation)
6. Eject the microSD card
7. Insert the microSD card into the microSD port on the back of the stock Ender3 v2 LCD panel.
8. Power on the Ender3 v2 and wait for the LCD to change from solid blue to solid orange.
9. After the display turns solid orange, turn off the Ender3 v2 and eject the microSD card from the slot on the back of the LCD panel.
10. Power the Ender3 v2 back on, observe the custom boot screen, then the main menu appears without any icons for the options & "Leveling" has been replaced with "Info"

**Expected behavior:** 
Custom icons/menus appear on the LCD screen

**Actual behavior:**
Blank icons on the menu screen and the "Leveling" option is replaced with "Info"

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue. -> [Google Photo Album](https://photos.app.goo.gl/ho3XGwpGVzyEUah79)

mriscoc, 2021-09-06:
Please, test with the last binaries released: https://github.com/mriscoc/Marlin_Ender3v2/releases/latest

Giadej, 2021-09-06:
@josurr rename the 9.ico file to 7.ico and try again.

josurr, 2021-09-07:
Thank you both for the assistance. Switching to the latest binaries corrected the issue.

:-)

closed 2021-09-07 by josurr