mriscoc/Marlin #116: 7x7 Mesh issue

eblieb, 2021-09-28:
Ender 3 V2 422 when I try to run a 7x7 mesh autolevel I get

Send: G29
[...]
Recv: //action:notification Probing Failed
Recv: //action:notification Printing...
Recv: X:215.67 Y:178.67 Z:6.60 E:928.92 Count X:17238 Y:14336 Z:2636
Recv: //action:prompt_end
Recv: //action:prompt_begin G29 Retrying
Recv: //action:prompt_button Dismiss
Recv: //action:prompt_show
Recv: //action:probe_rewipe

It works fine on 5x5

mriscoc, 2021-09-28:
I got that report with people using a CRTouch and not the BLTouch nor 3DTouch. What autolevel sensor are you using?

eblieb, 2021-09-28:
CRTouch, but it also did it with my BLTouch before I switched it. The CRTouch has been a lot more accurate and reliable than the BLTouch ever was.

eblieb, 2021-09-28:
I ran it with a 6x6 and it ran fine, so it is something about a 7x7 specifically.

mriscoc, 2021-09-28:
It always fail at the same point in a 7x7 mesh?

eblieb, 2021-09-28:
It fully completes all points and then when it is calculating the grid that is when it fails and starts over.

mriscoc, 2021-10-03:
After checking it seems that the CRTouch is not compatible with the HS mode, feed rate and the timings that this firmware uses for large mesh. The next version will have an option to disable the HS mode to be more compatible with this level sensor.

mriscoc, 2021-10-06:
Fixed in the last [experimental version](https://t.me/Ender3v2_Firmware/15757)
What's New in experimental 20211006 (compared to current Github binary releases)
* Probe Stow/Deploy in Probe Settings menu
* Probe offset XY maximums increased (again) to 60mm
* Mesh points increased up to 9x9
* Option to disable HS mode for better compatibility with CRTouch
* Option to change the Z Probe feed rate
* Home submenu with individual axis homing
* Enabled user setting for 115200 bauds (Creality Box compatibility)
* Rebuild power loss recovery user interface
* Head parking before recovery after power failure
* Maximum hotend temperature increased to 300°C for Volcano version
* New g-code C108 to cancel screens waiting for user simple confirmation
* Fix bug when exit from Info screen

closed 2021-10-10 by mriscoc