mriscoc/Marlin #117: M112 Error Ender 3 v2 stock connected to Octoprint

drak42, 2021-09-29:
Hi,

I have installed this firmware and all seems to be working well excpet for when connected to a pi running octoprint. I Have made all the changes specified in the installation guide but the printer keeps throwing M112 and shutsdown.

Any thoughts?

mriscoc, 2021-09-29:
Please follow this guide https://github.com/mriscoc/Marlin_Ender3v2/wiki/Octoprint

closed 2021-09-29 by mriscoc

drak42, 2021-09-29:
Hi, thanks for the quick response, I have followed that though and still get the issue

mriscoc, 2021-09-29:
Do you change the error handling in Octoprint? https://github.com/mriscoc/Marlin_Ender3v2/wiki/Octoprint#error-handling

jcass77, 2021-09-29:
I have encountered this error as well on the last few versions.

I have a BLTouch installed and in my case I need to initiate bed leveling from the printer the first time after the firmware is upgraded. 

I can then print and update the mesh from OctoPrint (using the the [Bed Level Visualizer](https://plugins.octoprint.org/plugins/bedlevelvisualizer/) plugin) from there on out without a M112 occurring.

mriscoc, 2021-09-29:
> I have encountered this error as well on the last few versions.
> 
> I have a BLTouch installed and in my case I need to initiate bed leveling from the printer the first time after the firmware is upgraded.
> 
> I can then print and update the mesh from OctoPrint (using the [Bed Level Visualizer](https://plugins.octoprint.org/plugins/bedlevelvisualizer/) plugin) from there on out without a M112 occurring.

That is because you have a `M420 S1` on your g-code file without a valid mesh.

drak42, 2021-09-29:
Hi,

 

Yes, followed the guide as instructed for M112 error handling

 

 

Thanks

Chris

Hr46ph, 2022-01-10:
There are suggestions but no solutions for me in this thread. Im running into this now as well. I had a few succesful prints and even one earlier today. Just now I am getting this error on a newly uploaded file to octoprint.

I verified all my settings, the error handling, there is a valid mesh and bed level visualizer is able to refresh and display it and its configured to be used in the starting gcode in Prusa slicer.

The earlier print (small pillar) still prints, but when I want to start a bigger print I receive this same error.

I tried reslicing it with different settings and I made some minor changes via OpenSCAD and re-exported it to try slicing it again. This (larger but still simple design) won't print using this firmware.

I've printed it before on the stock Ender 3v2 firmware.

Any ideas please?

mriscoc, 2022-01-10:
M112 is a generic error, we need a log from Octoprint and/or the capture of the screen to know what happened. Also, could be useful if you provide the g-code that produce the error.