mriscoc/Marlin #142: Wrong bed temperature using Cura, 50 degrees, displayed 60 degeees

R0B3RDV, 2021-10-31:
1. [First Step]
Create a file in the latest Cura with bed temp. 50 %
2. [Second Step]
print via Octoprint to the printer (mine Ender 3 V2)
3. [Third step]
Read 60 degeees instead of 50

mriscoc, 2021-10-31:
Check the temperature for your first layer. If you have an auto leveling command (G29) on your start script, then the auto level will preheat the bed to the leveling temperature which is by default 60 Celsius, this temperature is the same that tha PID temperature and you can change it in the Bed PID settings menu.

R0B3RDV, 2021-11-05:
So it doesnt matter what you use as bed temperature in your slicer ?

R0B3RDV, 2021-11-05:
I mean, when I want my print to be sliced at 50 degrees instead of 60, i've choose 60, but 3 hours ?

mriscoc, 2021-11-06:
Let me to elaborate, the printer sets the bed temperature according to g-code command and settings in display, then the final bed temperature will follow the last command issued. Check you g-code and look for what is the last bed temperature command and remember also that a leveling command (G29) also set the bed temperature. For example, I use two different temperatures for the bed, hotter for the first layer, and the printer follow those temperatures.

closed 2021-11-06 by mriscoc