mriscoc/Marlin #152: Remaining time from the slicer

zoltanf, 2021-11-21:
### Description

When enabling the remaining time generation in slicer (command M73), the value is shown on the screen, but it jumps between the slicer and the calculated time. It would be great to show only the slicer estimated remaining time when it's present in the gcode.

### Steps to Reproduce
1. Check the option "Supports remaining times" under printer settings in prusa slicer or super slicer. 
2. Generate a gcode file and run the print. Observe the displayed remaining time. In gcode it is updated approximately every minute.

**Expected behavior:** [What you expect to happen]
Show slicer estimated remaining time, as it is usually more precise.

**Actual behavior:** [What actually happens]
Remaining time switching on display between the firmware calculated remaining time and the slicer remaining time from gcode.

#### Additional Information

Will add pictures showing this on a print progress.

mriscoc, 2021-11-23:
Thank you for your report, we will investigate about it.

zoltanf, 2021-11-24:
Printing via µSD card.

mriscoc, 2021-12-12:
The printer updates the remaining time in an independent way when the print is via µSD. The only solution that I see for this behavior is to disable the internal update of the remaining time when the first M73 R### is received.

zoltanf, 2021-12-16:
I think that is a reasonable solution.

closed 2022-01-18 by mriscoc