mriscoc/Marlin #158: Firmware Bed Leveling Error When Printing with Octoprint V20211121

AdamWP, 2021-12-01:
I have "M420 S1 Z2" in my Cura start g-code in order to make use of the bed level mesh.  With prior versions of the firmware I had no issues and things worked as expected, but with version 20211121 if I print with Octoprint it returns "your printers firmware reported an error failed to enable bed leveling" and the printer has to be turned off.  

If I remove the "M420 S1 Z2" it works but I assume that the bed leveling mesh isn't used, is that correct?

wovka88, 2021-12-03:
do you run a new mesh after update? is the mesh also saved?

You can configure ocroprint for error-handling:
https://github.com/mriscoc/Marlin_Ender3v2/wiki/Octoprint#error-handling

mriscoc, 2021-12-04:
Octoprint is warning you that you are trying to use a mesh but there isn't a valid mesh stored in the printer. Please do an auto bed leveling before to print and save the mesh by Control/Store Settings.

closed 2021-12-04 by mriscoc