mriscoc/Marlin #159: Timer issue

Ram-smith, 2021-12-01:
Hi the timer is showing an odd time read out with 3 digits in the minutes area.  Please see photo 
[image]

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

mriscoc, 2021-12-04:
Thanks for your report, please, can you elaborate the steps to reproduce the issue? How long was the real remain time? Are you printing from Octoprint?

mriscoc, 2021-12-12:
Because no more feedback was written, I'm going to close this issue.

closed 2021-12-12 by mriscoc