mriscoc/Marlin #166: Manual Leveling not working entirely

Saurabhhsrivastavaa, 2021-12-11:
### Manual Leveling not working entirely
The manual Leveling is not working for Right side of the bed. If you choose Front Right or Back Right, the head does not move nor probes the bed. The rest of the options are working fine... Just the right side is not accepted by the system. 

### Steps to Reproduce
Choose Back Right in the Manual Mesh mode. 

1. [First Step] irrespective of 2x2 or 3x3 or any other grid chosen, the front/back right option does not move the print head. 
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Saurabhhsrivastavaa, 2021-12-11:
Can we also rename the title as ENDER 3V2 427 CR TOUCH?

mriscoc, 2021-12-12:
Please, see the note 1 of: https://github.com/mriscoc/Marlin_Ender3v2/wiki/Manual-leveling-(leveling-bed-corners)-menu
>Note1: If your BLTouch doesn't go to some points, you have unreachable points because your probe/home offsets, try to increase the probe margin in the probe settings menu.

closed 2021-12-12 by mriscoc

Saurabhhsrivastavaa, 2021-12-12:
No. It doesn't help. The probe does not travel to right. I increased the margin to 100.even then it doesn't work.

Saurabhhsrivastavaa, 2021-12-12:
> 

Not helping. Tried but still does not move.
My bed size is defined as 220x220x220

XMIN 0
YMIN 0
XMAX220
YMAX220

mriscoc, 2021-12-15:
Did your printer an auto home correctly?

Saurabhhsrivastavaa, 2021-12-16:
Yes. Absolutely.