mriscoc/Marlin #174: Problem with probing on larger bed, using Creality 4.2.2 with V2 screen,MS HE/DD & BLtouch on a Ender 3 MAX, so it is a V2 with a larger bed size.

SmallTownSound, 2021-12-17:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

SmallTownSound, 2021-12-17:
If you set the be larger to accommodate the 300x300 bed,  it has probing problems on the LEFT side of the bed.  No matter how high you set  the probe offset.  The probe is on the left side of the nozzle, so it makes no sense to have probe issues on the left side of the bed, I could see issues on the right side, but nothing you set will make the left points work properlt.

mriscoc, 2021-12-18:
Ok, I don't have a bigger bed, so I had to release the belt tensioners to avoid crashes.
I did this test: load defaults, change only physical settings:
X max position: 300
Y max position: 300
X bed size: 300
Y bed size: 300
Went to Manual Leveling and select first front left and then back left. All movements were OK.
Please confirm.

mriscoc, 2022-01-07:
Is there any news regarding this issue?

closed 2022-01-13 by mriscoc