mriscoc/Marlin #179: Autolev Levelling Sensor

RPI-PMK, 2021-12-21:
### Discussed in https://github.com/mriscoc/Marlin_Ender3v2/discussions/78

<div type='discussions-op-text'>

<sup>Originally posted by **RPI-PMK** July 30, 2021</sup>
Hi,

I have an Autolev sensor from Substancia 3D and trying to get it working with your great firmware. Been using Manual Mesh for a while now and had no issues so far.

They use a modified Jyers firmware which they host on their site.

I have details of changes needed in Configuration.h and Configuration_adv.h but stuck as to how to make these changes to get it working. Can you please point me in the right direction ??

Thanks.</div>

mriscoc, 2021-12-22:
If you can create a json configuration file with your specific change I will try to review a special version, read here: https://github.com/mriscoc/Marlin_Configurations

RPI-PMK, 2021-12-22:
Not sure what a json file is.

I got this from supplier. How would I implement this in firmware??

Configuration on Configuration.h
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define FIX_MOUNTED_PROBE
#define X_PROBE_OFFSET_FROM_EXTRUDER 0 // X offset: -left +right [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0 // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0 // Z offset: -below +above [the nozzle]
#define AUTO_BED_LEVELING_BILINEAR
#define Z_SAFE_HOMING
#define MULTIPLE_PROBING 2
//#define MIN_SOFTWARE_ENDSTOP_Z
#define Z_CLEARANCE_DEPLOY_PROBE 3 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES 2 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE 2 // Z Clearance between multiple probes
//#define Z_AFTER_PROBING 5 // Z position after probing is done
#define Z_PROBE_LOW_POINT -2 // Farthest distance below the trigger-point to go
before stopping

Configuration on Configuration_adv.h
#define BABYSTEP_ZPROBE_OFFSET
#define DOUBLECLICK_FOR_Z_BABYSTEPPING

I have it installed on ender 3v2 and can measure the offsets but that is as far as I have got.

RPI-PMK, 2022-01-07:
@mriscoc any pointers for making a json file, I am complete noob on software so not sure where to start ??

Will have another look this weekend.

mriscoc, 2022-01-11:
I'm a little busy right now, but I want to put in my Configurations repository some specialized variants, including other probes and hotends.

RPI-PMK, 2022-01-11:
Thanks, that will be great. Printer still works with your manual mesh version for now. Let me if I can help.

closed 2022-02-04 by mriscoc