mriscoc/Marlin #180: Bed PID kd value

tome9111991, 2021-12-22:
Hello
can it be that the kd value in the bed pid menu is cut?

Display show me 343.27 but it should be 843.27.
Because in config i have #define DEFAULT_bedKd 843.27

Look at the picture looks firtst value is cuted by set kd
[image]

I use latest stock marlin bugfix version.

mriscoc, 2021-12-22:
This is a problem with your DWIN_SET some fonts are little larger and have alignment errors, please update your DWIN_SET using the stocks fonts file from here: https://github.com/mriscoc/Marlin_Ender3v2/blob/Ender3v2-Released/display%20assets/stock/DWIN_SET/0T5UIC1.HZK

tome9111991, 2021-12-22:
Good to know thanks 
But is there a way to fix with the stock marlin font,?

mriscoc, 2021-12-22:
> Good to know thanks But is there a way to fix with the stock marlin font,?

I didn't try, can you test with the original CrealityUI for see if a cut is happened there?

tome9111991, 2021-12-22:
Im not sure CrealityUI  had PID no menu.
But i will look later

mriscoc, 2021-12-23:
The current Marlin font also causes other failures, is better to install the stock Creality font that I linked.

tome9111991, 2021-12-28:
I have flashed with your font.
The value looks correct now.

I will test the font with the other gui's if this work correct maybe i create a Pull request with the new font.

mriscoc, 2021-12-28:
Thanks for the confirmation

mriscoc, 2021-12-29:
> I have flashed with your font. The value looks correct now.
> 
> I will test the font with the other gui's if this work correct maybe i create a Pull request with the new font in marlin configuration repo.

If you have access to the Marlin Discord you can see this issue reported before: https://discord.com/channels/461605380783472640/491105274464043026/912038168835006494

The-EG, 2021-12-29:
> noted that the original Creality font has an eight with only 9 pixels width, while the Marlin font has an eight with 10 pixels width for the case of font12x24

This is still not true. The fonts are displayed at the same width regardless of what glyphs they contain. They are both 12 pixels wide in that case, it's just the character only takes up 8 or 10 of those pixels. However, whenever the LCD draws a string, each character is given the full width. If you draw a string that is 2 characters wide in 12x24 font, it's going to use 24 pixels of width to do it. It doesn't matter which font you use, that's how fixed width fonts work.

mriscoc, 2021-12-29:
> > noted that the original Creality font has an eight with only 9 pixels width, while the Marlin font has an eight with 10 pixels width for the case of font12x24
> 
> This is still not true. The fonts are displayed at the same width regardless of what glyphs they contain. They are both 12 pixels wide in that case, it's just the character only takes up 8 or 10 of those pixels. However, whenever the LCD draws a string, each character is given the full width. If you draw a string that is 2 characters wide in 12x24 font, it's going to use 24 pixels of width to do it. It doesn't matter which font you use, that's how fixed width fonts work.
> 
> I was explaining why one font may appear smaller than another, but I'm reality all fonts of a given size are always that size.

The grid is the same, but the pixels (count the black boxes in the drawing) that the digit currently uses are different.

mriscoc, 2022-01-13:
No new feedback

closed 2022-01-13 by mriscoc