mriscoc/Marlin #187: Firmware reboots when selecting Print immediately on power on

net147, 2022-01-01:
### Description

Selecting Print immediately on power on as soon as it appears causes the firmware to reboot. I have also seen it happen sometimes during normal operation.

### Steps to Reproduce

1. Turn on printer
2. Click the knob to select Print as soon as it appears

**Expected behavior:** Shows the list of available prints

**Actual behavior:** Firmware reboots

#### Additional Information

* Firmware: Ender3v2-422-BLTouch-20211121.bin
* Micro SD card is inserted

mriscoc, 2022-01-02:
Format your µSD card of change it to a faster one.

net147, 2022-01-02:
> Format your µSD card of change it to a faster one.

It is the stock micro SD card that came with it with all the folders/files deleted and a couple of GCode files added to it. Have you seen it help formatting it from Windows again or changing to new card?

I have also seen it reboot going back into the Print menu after the previous print finished.

mriscoc, 2022-01-02:
The card that comes with the printer is not always a good quality one.

net147, 2022-01-02:
Okay, will try it later today. It never happened with the Jyers E3V2-BLTouch-3x3-HS-v4.2.2-v2.0.1.bin firmware. It is the first time I have seen it after switching to this firmware.

closed 2022-01-03 by mriscoc