mriscoc/Marlin #199: M600 Filament change doesn't return print head nor extruder relative position mode

vercas, 2022-01-09:
### Description

After resuming the print from a filament change, the print head doesn't move back where it was.
Additionally, it resets extruder positioning to absolute mode.

### Steps to Reproduce

 1. Introduce a layer change in a slicer that uses relative E positions (like PrusaSlicer), e.g. for one of these models: https://www.thingiverse.com/thing:4584438/files (they print quickly)
 2. Change filament when it pauses
 3. Resume print

**Expected behavior:** Printer returns all 3 axes to the position they were before and it resumes printing.

**Actual behavior:** Printer keeps going without restoring the position nor the extruder relative mode. PrusaSlicer will make it move the printer to the X and Y coordinates where it needs to start the layer, but *not* the Z position, as that is done before the M600.  
Additionally, it will squirt a large blob of filament and attempt to continue printing. During this attempt, it will only leave tiny blobs behind:  
[image]
[image]

#### Additional Information

I think the extruder switching to absolute position is an older bug in Marlin, because I was seeing exactly the same issue with Creality's stock firmware when using M25 + a lot of manual gcode to do a filament change:
[image]

mriscoc, 2022-01-10:
M600 saves the actual position and moves the print head to the parking point, then it unloads the filament and waits for user input, after confirmation, the extruder loads filament and does a purge, asking to the user to continue for return to the saved position to continue printing or purges again, that behavior can trigger the change to the absolute positions. The firmware implements the ADVANCED_PAUSE_FEATURE and PARK_HEAD_ON_PAUSE

vercas, 2022-01-10:
I've only resumed the prints from Octoprint so far, where I don't have any option except `Resume`. I didn't even look at the screen, if the option is supposed to be there. The firmware certainly didn't load filament and purge on its own. I did that manually before I resumed the print. It 100% does **not** attempt to move the print head back where it was.  
I won't be able to test upstream Marlin in the short term. I've only got one printer and it's quite busy right now.  
I wonder if the problem here is that I'm resuming from Octoprint...

vercas, 2022-01-11:
Issuing an `M600` doesn't make any prompt appear on the LCD or in Octoprint.  
If I use the LCD to go to the Tune menu and select Change Filament in there, then I finally see the options you are talking about.

mriscoc, 2022-01-11:
> Issuing an `M600` doesn't make any prompt appear on the LCD or in Octoprint. If I use the LCD to go to the Tune menu and select Change Filament in there, then I finally see the options you are talking about.

If you can't see the options using Octoprint, then the M600 is being stripped.